rusefi/hellen-NB1-issues #3: What PCB outline do we use?

rusefillc, 2021-07-02:
2000 miata OEM case looks like NB2 case with 64 pin connector

2000 miata OEM does NOT look like NA6 or Frankenso case

[image]

rusefillc, 2021-07-02:
2000 camry looks like the winner to me

1) it looks to match NB1/NB2 OEM style so brackets might align

2) it looks to use a wide 125 connector OEM so we would be able to fit 76 pin connector

[image]

rusefillc, 2021-07-02:
[image]

rusefillc, 2021-07-02:
@stefanst says that 2000 camry case would give us a mechanical interference with one of the Miata OEM ecu mounts?

stefanst, 2021-07-02:
Fortunately, my memory was wrong. It looks like there won't be interference. The bracket mounts to the rear of the PCB

rusefillc, 2021-07-08:
2000 camry case has arrived it totally works for us!

but two not one PCB to case mount and different location, more info later

rusefillc, 2021-07-09:
Dimensions notes see pictures

[image]

[image]

[image]

rusefillc, 2021-07-09:
2000 camry case is very available, let's make NB1 board compatible ONLY with 2000camry case not with 1996camry

meaning we need two new mounting holes but we can remove existing mounting hole from the 1996camry location

closed 2021-08-09 by rusefillc