rusefi/hellen-NB1-issues #4: 12 pin Options connector

BeerMoneyMotorsports, 2021-07-02:
x8 pins dedicated to wbo leaves 4 pins available.  

1- Temperature input
2- Digital input
3- Boost Control
4- VVT  
5- +5v
6-  Sensor Ground

wbo
7- Heater -
8- Heater +
9- Vs
10- Vs/Ip
11- Ip
12- R_Trim

BeerMoneyMotorsports, 2021-07-02:
1- PA1
2- PF3
3- PF12 with diode to 12v
4- PF14 with diode to 12v
5- 5v
6- tie to 4c/f

andreika-git, 2021-07-16:
1) We want "3-Boost Control" output, and we already have pin 3T for "EGR Boost Sensor Solenoid Valve" which currently uses INJ6 (PF12). Do we want to re-use 3T pin for boost control (does anyone need EGR at all?) ? Or use IGN6 for a new pin on 12-pin connector and leave 3T pin uncontrolled (free pad)?
2) The same goes for VVT - we already have pin 3Q for "VICS Solenoid" which uses INJ8 (PF14). Do we want to re-purpose this pin for VVT or move it to 12-pin connector and leave 3Q unconnected?

BeerMoneyMotorsports, 2021-07-16:
1- repurposing it for boost is acceptable. In my MS days we used 2L for boost since no one cares about MAF. 

2- i would like to see VICS remain as it does improve power. See dyno below. 

[image]

andreika-git, 2021-07-16:
Ok, then maybe we could use OUT_INJ5 for VICS solenoid. INJ5 is currently occupied by OUT_AC_FAN_RELAY (legacy of NA6 design), but we can use an ordinary OUT_LOWxxx for relay.

JimRowan, 2021-07-16:
maybe I misunderstand, but this makes it non-PNP, right?  harness changes are required?

BeerMoneyMotorsports, 2021-07-16:
@JimRowan this would maintain PNP and repurpose unused connectors for auxiliary functions.

andreika-git, 2021-07-16:
We can use jumpers and connect the BOOST output to both OEM 3T (DNP by default) and aux.port (jumper is soldered by default) pins. So that a user can install a boost solenoid without altering the OEM harness. Also if we implement EGR support someday, it would be possible to enable the OEM EGR boost solenoid (3T) by switching the jumpers on board.
[image]

BeerMoneyMotorsports, 2021-08-09:
@andreika-git I like this pinout.  Please continue with this as our Options Port pinout. 
[image]

andreika-git, 2021-09-09:
Done!
[image]

closed 2021-09-09 by andreika-git