rusefi/hellen-NB1-issues #9: ETB controller

BeerMoneyMotorsports, 2021-07-27:
Reference below.  Add ETB functionality and board connector for I/O.  

https://github.com/rusefi/hellenNA6_issues/issues/14

BeerMoneyMotorsports, 2021-08-02:
@andreika-git add on etb driver and vertical board mounted connector. Connector requires 10a capacity.

rusefillc, 2021-08-09:
@andreika-git does https://github.com/rusefi/hellenNA6_issues/issues/14#issuecomment-895500346 have all the details needed?

rusefillc, 2021-08-11:
@andreika-git says that we are out of space so one 6 pin connector is preferred for both TPS and PPS sensors and power. i.e. one shared GND for all four sensors and one shared +5v for all four sensors

rusefillc, 2021-08-11:
@BeerMoneyMotorsports we are starting to hit the limit of 16 analog inputs total 
see #10

@BeerMoneyMotorsports please link a google spreadsheet listing all analog inputs on NB1 it looks like we would have to start sharing. proposal is for optional ETB channels to overlap with optional  IN_PRESSURE, IN_TEMP, IN_EGR_BOOST

another option is to start looking into external ADC but that's a separate more complex story

andreika-git, 2021-09-09:
Done!
[image]
[image]

closed 2021-09-09 by andreika-git