rusefi/hellen-NB1-issues #10: AC pressure is it digital or analog?

rusefillc, 2021-08-11:
@BeerMoneyMotorsports did we have some AC pressure drama on some other miata?

please link relevant tickets

please confirm which way is NB1 digital or analog

this is all related to analog input shortage if we go with ETB option

BeerMoneyMotorsports, 2021-08-13:
https://github.com/rusefi/hellen64_miataNA6_94/issues/52

**Chatting with @BeerMoneyMotorsports we've figured out why we're seeing what we are (it's the diodes - it can't pull to ground, it can only pull to ground plus two diode drops).

@Qwerty-OFF Let's just route AC switch input to IN_AUX3. That has an integrated pullup, plus the divider gets us levels that might actually work as a plain digital input. If not, we can do it analog instead.**

@mck1117 will have to chime in on if it will be possible to utilize the digital input alone.  

[image]
NB1 is identical to NB2 for AC switch input.  We can utilize one of the dual type inputs like below. 

 - pin: 4F
    id: [GPIOB_0, EFI_ADC_8]
    class: [switch_inputs, analog_inputs]
    ts_name: 4F - AC_PRES/AUX1 (A23)
    function: A/C Pressure In

andreika-git, 2021-09-09:
Done!
IN_AC_SW is on IN_AUX3:
[image]

closed 2021-09-09 by andreika-git