rusefi/hellen-NB1-issues #11: Schematic Review Notes

BeerMoneyMotorsports, 2021-08-19:
@andreika-git i have reviewed the schematics and they look great.  I do have a few notes below.  

1G- In_Steering -  This can be deleted and the input can be repurposed
1Q- Needs to be Out_Alternator Warning
2A- Fuel Tank Pressure Sensor - not needed and input can be repurposed
2G- IN_AFR - Use 2C for IN_AFR
3D- IN_MAP - Shows unused on chassis schematics?  Is it for a EU miata?

andreika-git, 2021-08-28:
1G, 2A - What do you mean by "can be repurposed" and "can be deleted"? We cannot just delete the input if we want to use it in a different way. And we already allow to repurpose any input - it's up to a user to decide what to do with his harness, right?
1Q - done
2G - we already have IN_O2S on that pin, and IN_AFR is just another one (like a second lambda). The firmware allows to select any of them, there is no difference, just different names.
3D - that's just a non-OEM pin to use as an external MAP sensor (if a user wants to) - like 4U pin on Hellen72

BeerMoneyMotorsports, 2021-08-28:
1G and 2A are not needed for the NB1. We dont use either of those inputs.  If we are running low on inputs, we can delete them and use the stm pin for something else. I agree with your other notes.

closed 2021-09-22 by BeerMoneyMotorsports