rusefi/hellen-NB1-issues #12: JLC-BOM shortage

BeerMoneyMotorsports, 2021-09-14:
Below are the components missing from JLC.  @rusefillc can you give some input on options for 1 and 4?  @andreika-git can you take care of 3 please?

1- U5 | L6205PD | SO20POWER | L6205PD013TRPowerSO-20 Motor Driver ICs ROHS
2- F1,F2,F3 | Fuse-Pad-Pad | PAD-1206-PAD | 12H1500D6.3 - 100A 5A Chip Fuse - - SMD1206 Surface Mount Fuses ROHS
      **C553924 as a replacement?**
3- U105 | STM32F407IGT6 | LQFP176 | STM32F407IGT61.8V ~ 3.6V 32-Bit ARM® Cortex®-M4
      **Please convert to 144 pin STM32F4**
4- L420 | NLV32T-2R2J-PF | 1210 | NLV32T-2R2J-PF1210 Inductors (SMD) ROHS

BeerMoneyMotorsports, 2021-09-14:
1- looks like mouser has plenty if i need to solder by hand.  https://www.mouser.com/ProductDetail/STMicroelectronics/L6205PD013TR?qs=%2Fha2pyFaduju3fz9SGs7CYPcSItrN5EFuA%252B4MopebmQoCB3sRejAuQ%3D%3D&utm_source=octopart&utm_medium=aggregator&utm_campaign=511-L6205PD013TR&utm_content=STMicroelectronics

rusefillc, 2021-09-14:
overall it's not my strongest field :( 

1 we can redesign for another driver chip I guess but maybe simpler to hand-solder?

4 C89016 but low stock

andreika-git, 2021-09-15:
My suggestions:
1 - x
2 - C553924
3 - C1337111 (STM32F407ZGT7 144pin 1Mb 192kb)
4 - C223413 (plenty of stock)

andreika-git, 2021-09-15:
Also we need to revisit the "#non-JLC components" part of the BOM.
For example, we have a JLC replacement for the injector mosfets (Q305-Q312) back in stock: ZXMS6005SGTA_C174042 - which was already used in Hellen boards. Also U103 and U480 are back in stock.

andreika-git, 2021-09-15:
I've updated the gerber & BOM files on the repo, please try this version.

mck1117, 2021-09-15:
does switching to lqfp144 stm32 cause any pin incompatibility? does the default config reference any pins that change between the 176 and 144?

andreika-git, 2021-09-15:
> does switching to lqfp144 stm32 cause any pin incompatibility? does the default config reference any pins that change between the 176 and 144?

Yes, some pins are different. For example, 144 doesn't have port I, so some of the OUT_IGNx and OUT_IOx pins are different. I tried to maintain the compatibility for all remaining pins. 
In a nutshell, the mcu144 requires a different firmware "yaml" config. We don't have a yaml for NB1 yet, so nothing to worry about.

mck1117, 2021-09-15:
OK - so long as it's for a board that has none "in the wild", then that's no problem!

BeerMoneyMotorsports, 2021-09-15:
We have a new part shortage showing with the latest BOM.

Y101 | 32.768 kHz | SMD2012 | FC-12M32.768KHZ7PF20PPM-40℃~85℃ 32768Hz 贴片无源晶振 ±20ppm 90kΩ 7pF SMD2012-2P Crystals ROHS

andreika-git, 2021-09-15:
Can you manually edit the BOM file and replace the Y101 with C97602?
"32.768 kHz","Y101","SMD2012","C97602"

rusefillc, 2021-09-22:
Kind of related to https://github.com/rusefi/hellen-NB2-issues/issues/13

closed 2021-09-22 by BeerMoneyMotorsports