rusefi/hellen-NB1-issues #15: critical error pin

BeerMoneyMotorsports, 2021-09-28:
firmware is giving following error.

2021-09-28_11_53_20_485: EngineState: CRITICAL error: led: comm status: startup pin state PH10 actual value=0 logical value=1 mode=OM_INVERTED 20210928@hellenNB1

firmware: 20210929@28919

rusefillc, 2021-09-28:
GPIOH_10 is 176 pin comm pin

please use TC2030 to fully erase chip

please use manual DFU to flash latest 144 pin firmware bundle from snapshot

please report back

BeerMoneyMotorsports, 2021-09-28:
Performed full chip erase via STMCube programmer.  loaded latest firmware.  no change.  

https://www.rusefi.com/build_server/

BeerMoneyMotorsports, 2021-09-28:
2021-09-28 11:37  from above url. hellen-nb1

BeerMoneyMotorsports, 2021-09-29:
https://rusefi.com/build_server/rusefi_bundle_hellen-nb1.zip  direct link to build used.  2021-09-28 11:37

rusefillc, 2021-09-29:
I am out of ideas :(

@andreika-git can you have a second look?

rusefillc, 2021-09-29:
code defect fixed, relevant code refactored to reduce copy/paste and reduce chance of similar issue in the future

closed 2021-09-29 by rusefillc