rusefi/hellen-NB1-issues #23: vvt target table

BeerMoneyMotorsports, 2021-11-02:
we have an aggressive table and we need to set it more like other known good tables.

ours
[image]

other known good
[image]

rusefillc, 2021-11-03:
bad ticket

1) images instead of tune URL
2) vague. could not not be "something like" has to be 100% exact.

BeerMoneyMotorsports, 2021-11-03:
1- images as i wasnt able to upload the default settings from my ms3.  https://www.megasquirtpnp.com/downloads/basemaps/fw_pro_132/dlmap.php?dlfn=MM0105_Miata_132.msq

2- make it exactly like this.

rusefillc, 2021-11-03:
@BeerMoneyMotorsports please provide a URL for the table you would like to be hard-coded as default and I will make it the default!

BeerMoneyMotorsports, 2021-11-03:
fuck, this was supposed to be nb2, but i suppose it should be the default for any miata vvt table.  the url i posted is to the tune with the needed settings.