rusefi/hellen-NB1-issues #25: Alternator improvements

BeerMoneyMotorsports, 2021-11-29:
Miata OEM ecu uses a 1k resistor in series plus a 100nf cap to ground. We meed to match this in our circuit. 

[image]

andreika-git, 2021-11-29:
[image]

BeerMoneyMotorsports, 2021-11-29:
@rusefillc yes, this ticket will be added to the main nb2 update list. 

https://github.com/rusefi/hellen-NB2-issues/issues/21

BeerMoneyMotorsports, 2021-11-29:
@rusefillc yes, this ticket will be added to the main nb2 update list.

rusefillc, 2021-11-30:
@BeerMoneyMotorsports I am happy to help

[image]

rusefillc, 2021-12-03:
Status update: NB1 rev B being fabricated with new MCU module

andreika-git, 2021-12-03:
...with new Output-0.3 module

BeerMoneyMotorsports, 2021-12-14:
Oliver has installed the resistor and cap upgrades several weeks ago.  To date, he has had no issues.  I will poke him for an update

mck1117, 2021-12-14:
he mentioned on slack a few days ago that he's noticed no performance difference with the added RC, so we should be good to make the change

BeerMoneyMotorsports, 2021-12-14:
Oliver reports 50kms and no issues

BeerMoneyMotorsports, 2021-12-14:
the next batch, arriving tomorrow, have the new module and the alternator updates

BeerMoneyMotorsports, 2022-01-07:
new module works as desired.  no faults yet.  will close now.

closed 2022-01-07 by BeerMoneyMotorsports