rusefi/hellen-NB1-issues #26: BOM Shortages 11/30/21

BeerMoneyMotorsports, 2021-11-30:
BOM shortages for NB1.  Please assist in alternatives.

1- U420 -- LMR14030SDDAR -- C182078
2- Y101 -- FC-12M32.768KHZ7PF20PPM -- C99009
3- U410 -- MIC5205-3.3YM5-TR -- C37970
4- Q601 -- VNP28N04-E --  C263002

andreika-git, 2021-11-30:
1 - U420 - ??? we need @mck1117's help here, I don't know what's the best replacement for "his" LMR140x0...
2 - Y101 -- C97602
3 - U410 -- C129997
4 - Q601 -- C501116 (VNP20N07-E) *MAYBE*, I'm not sure...

BeerMoneyMotorsports, 2021-11-30:
@mck1117 can you offer some input please?

mck1117, 2021-12-01:
1. Hmm, not an easy replacement for this that I can see. In stock at mouser, though: https://www.mouser.com/ProductDetail/Texas-Instruments/LMR14020SSQDDAQ1?qs=sGAEpiMZZMsyYdr3R27aV1%2FsE%2F4eX1bpBt4%2FHWGByuc%3D
4. C501116 sounds good - maybe even better than the original did. Why do you say "maybe," @andreika-git?

andreika-git, 2021-12-01:
> 2\. C501116 sounds good - maybe even better than the original did. Why do you say "maybe," @andreika-git?

I just didn't have time to check the datasheet more in depth.

BeerMoneyMotorsports, 2021-12-01:
thank you @andreika-git and @mck1117

closed 2021-12-01 by BeerMoneyMotorsports