rusefi/hellen-NB1-issues #29: Alternator charging problem

BeerMoneyMotorsports, 2022-02-21:
**Board**- NB1 rev B
**Log**- https://rusefi.com/online/view.php?log=640

Problem-  Alternator will stop charging and voltage will drop to ~12v.  It will stay there a few seconds and then it will ramp up to 17-18v before coming back to normal 14.2v.  

The log indicates that the ECU is indeed requesting the activation of the alternator but it is not responding.  

I have changed the alternator, reviewed chassis wiring, tried another Hellen.  No improvements.  I have also tried several settings in the tune to correct with no improvements.  

I **will be testing alternator output wire with an Oscilloscope today.**   I started by bench testing with the scope.  ecu had 12v supply and was stimulated to simulate 1200rpm.
[alternator bench test.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8112146/alternator.bench.test.pdf)

BeerMoneyMotorsports, 2022-02-21:
[alternator idle.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8112606/alternator.idle.pdf)
[alternator idle high load.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8112609/alternator.idle.high.load.pdf)
[alt reved 12v.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8112612/alt.rev.12v.pdf)
[alternator 17v.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8112617/alternator.17v.pdf)

BeerMoneyMotorsports, 2022-02-22:
circuit tests
Alternator control(1o) wire to ground
ECU disconnected- 1.026 kohm
ECU and alternator disconnected- OL

BeerMoneyMotorsports, 2022-02-22:
[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)  [7:36 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490212114379)
can you measure alt pin -> ground on the ECU?
[7:36](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490215172999)
unplugged from car
[7:38](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490282592919)
are either of R309 or R310 installed?
[7:38](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490310461799)
I have a hunch neither is installed

[BeerMoneyMotorsports](https://app.slack.com/team/U01JUH5H6EN):speech_balloon:  [7:38 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490314242429)
OL
[7:43](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490599050929)
I believe r310 is in place.
[7:44](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490691092599)
I dont see r309 on ibom

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)  [7:45 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490724637339)
OK, I expect R310 to be present and R309 missing
[7:45](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490730800099)
can you measure R310?
[7:45](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490732764369)
should be 0

[BeerMoneyMotorsports](https://app.slack.com/team/U01JUH5H6EN):speech_balloon:  [7:48 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490934363359)
Give me 15

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)  [7:49 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645490965979339)
and check the voltage both sides of R310 with ECU connected/ign on/whatever

[BeerMoneyMotorsports](https://app.slack.com/team/U01JUH5H6EN):speech_balloon:  [8:03 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645491838119039)
can i do that on the bench?  its a royal pain in the ass in the car
[8:04](https://rusefi.slack.com/archives/C0323M3MPU4/p1645491858017629)
i dont have an ibom for nb1-b
[8:04](https://rusefi.slack.com/archives/C0323M3MPU4/p1645491873772269)
i need some education on how to get that
[8:04](https://rusefi.slack.com/archives/C0323M3MPU4/p1645491892349969)
its not on https://rusefi.com/docs/ibom/

[andrey](https://app.slack.com/team/U4PVDN5L7)  [8:07 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492031597889)
https://github.com/rusefi/hellen-NB1/tree/main/boards/hellen64_NB1-a/board has ibom
New

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)  [8:07 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492052666479)
Bench OK if power supply the same as in a car

[andrey](https://app.slack.com/team/U4PVDN5L7)  [8:07 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492054357519)
https://github.com/rusefi/hellen-NB1/tree/main/boards/hellen64_NB1-b/board does not
[8:07](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492062693989)
weird worth poking [@andreika](https://rusefi.slack.com/team/U4YULLW6Q)

[BeerMoneyMotorsports](https://app.slack.com/team/U01JUH5H6EN):speech_balloon:  [8:08 PM](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492113680859)
that will have to be tomorrow then.  that part of the board had changes from A to B
[8:11](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492290519869)
[@Matthew Kennedy](https://rusefi.slack.com/team/U80K9BYVD) r310 shows vbatt on both sides
[8:11](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492312799999)
warning, i am assuming this is r310.  things moved around from A to B
[8:12](https://rusefi.slack.com/archives/C0323M3MPU4/p1645492331011519)
it isnt in the exact same spot but very close

BeerMoneyMotorsports, 2022-02-22:
awaiting ibom for confirmation of location of r309 and r310.  Poked @andreika-git for support with ibom for NB1 rev-B

BeerMoneyMotorsports, 2022-02-22:
@mck1117 please advise what the expected voltage output should be on the oscilloscope readout.

mck1117, 2022-02-22:
I would expect to see:

1. 12v on both sides of R310
2. 12v on pin 6 of U302
[image]
3. PWM at 0v/12v on U302 pin 7 and on the alternator output pin.

andreika-git, 2022-02-22:
[image]
R310 is the one with "12" label.

BeerMoneyMotorsports, 2022-02-22:
[image]

BeerMoneyMotorsports, 2022-02-27:
Pin 6 u302= 12v
Pin 7 u302= 12v pwn(on bench)

is there a good way to simulate it being on the car?  Connecting alternator output to some load maybe?

BeerMoneyMotorsports, 2022-03-01:
hellen nb1 rev A shows different performance. No charging issues.

[nb1_a_alternator.pdf](https://github.com/rusefi/hellen-NB1-issues/files/8164810/nb1_a_alternator.pdf)

andreika-git, 2022-03-01:
The main difference between rev.A and B is this:
[image]
Maybe try to replace R23 with a jumper (0 ohms resistor)?

mck1117, 2022-03-01:
[image]

Rev A, alternator disconnected. This looks like the tc4427 doesn't have power at all.

BeerMoneyMotorsports, 2022-03-01:
@andreika-git there were also changes to the output module?  added some circuit protections s well?

BeerMoneyMotorsports, 2022-03-02:
@mck1117 @andreika-git I have retested NB1 rev A.  Previous test that showed only ~2v was due to user error.  rev B is still only ~2v.

[image]

BeerMoneyMotorsports, 2022-03-05:
i was able to probe the 4227 while on the car.  i found that at pin 7(output), i was getting a nice battery voltage pwm signal.  Exactly as expected.  I then bypassed R23(1k) and R312(100) and retested.  we then had 11.1v pwm and great alternator control.    It was determined that the alternator resistance must change during operation.  I removed R23 and left R312.  ~4.5v pwm and good alternator control.  so the problem was a result of the double resistors on the output.  We need to replace R23 with a 0ohm resistor or delete from circuit.  i am creating tickets for it.  this can be closed.

closed 2022-03-05 by BeerMoneyMotorsports